ArduPilot/ardupilot_wiki #3461: Rover: document recent sailboat features

arikrupnik, 2021-04-18:
@Hwurzburg : listening to your DevCon talk inspired me to send you some bullet points.

Sailboat support has grown since the last Wiki update.

* Sailboat supports auxiliary motor propulsion (`RCx_OPTION=74 Sailboat motoring 3pos`)
  *  `USE_MOTOR_NEVER`
  * `USE_MOTOR_ASSIST`
    * for tacking
    * for low wind
  * `USE_MOTOR_ALWAYS`
* Throttle input can be used to control sails, or the dedicated `RCx_OPTION=207 MainSail`. It would be helpful to explain the difference between them.
* Throttle input controls sails directly in manual mode. In Acro mode, throttle controls speed, in the sense that below a certain throttle value, the boat will depower the sails to remain stationary.
* Wiki describes only "traditional" sail control with a sheet and a winch.
  * An additional configuration is supported in release code--wing sail with elevator-style AoA control (`SERVOx_FUNCTION=128 Wing Sail Elevator`) https://discuss.ardupilot.org/t/sailboat-support/32060/321
  * A third configuration is WIP, direct servo control of rotating mast: https://github.com/ArduPilot/ardupilot/pull/17066/

Hwurzburg, 2021-04-18:
super...I may ask for more details when I get to this...

arikrupnik, 2021-04-18:
Thank you, @Hwurzburg. I'm sure you know this, but @IamPete1 is the authority on sailboat support. I'm just figuring things out.

Another aspect you may want to document is land sailing. The Wiki notes, "Only sail boats have been tested so far, although there is no reason land based vehicles wouldn’t work." Since this statement, @srmainwaring has had significant, documented success with land yachts.

PR #3561 (2021-05-31): Rover: Sailboat: update
@arikrupnik @srmainwaring I would be great full if you could have a quick look over this and make sure its mostly coherent., Thanks!

Fixes #3461
closed 2021-05-31 by Hwurzburg