ArduPilot/ardupilot_wiki #3585: Precision Docking

Cadify, 2021-05-31:
[Search](https://ardupilot.org/dev/search.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

nkruzan, 2021-05-31:
Hello do you have some information to add why there is an issue? I see you are interested in Precision Docking, but what is missing or needing clarification?

Cadify, 2021-05-31:
Hello,

 

Sorry, that was just a clumsy click by me, not intentional, other than that that I am running an open source project with ArduBoat where precision docking and maneuvering is very interesting, so I am looking into anything that may be relevant.

 

Actually, I was looking for some status update about how the Precision Docking is going. 

 

Regards

Joern Watvedt

RC boat joystick piloting project manager.

nkruzan, 2021-05-31:
I see it on the roadmap under @rmackay9 he may be able to give status update.

khancyr, 2021-05-31:
We are working on releasing the version v4.1, so it would surely happen after the release. If that isn't an issue with the wiki, I am closing this.

closed 2021-05-31 by khancyr

rmackay9, 2021-05-31:
Yes, I'd like to get precision docking working but I can't promise when yet I'm afraid..