ArduPilot/ardupilot_wiki #3596: High Definition Live Video FPV Systems

AndreHenke, 2021-06-08:
I am a little confused about Ardupilot and the DJI air unit: Do I need a special arduino converter in order to get OSD or can I connect it directly to a pixhwak serial port? What is the pinout diagram? 
Thank you!

Hwurzburg, 2021-06-08:
sorry,this is not for support issues....in the future use discuss.ardupilot.org
that said, no additional converter needed...connect it to any serial port, and set the protocol to 33 (DJI FPV)....you can setup the layout in MP just like you do with boards with integrated OSD...read https://ardupilot.org/plane/docs/common-msp-osd-overview.html

closed 2021-06-08 by Hwurzburg

AndreHenke, 2021-06-08:
Thanks for your help. I will try at discuss.ardupilot.org
However I cannot find MSP_OPTIONS = 0 (polling mode) on my pixhawk...
Thank you

Hwurzburg, 2021-06-08:
must use "latest"firmware for DJI support

Hwurzburg, 2021-06-09:
1) the firmware must be 4.1 ("latest")
2) the 1MB boards like Pixhawk1 may not include that feature since flash space is limited....but an inexpensive autopilot like MatekF765 has enough flash

AndreHenke, 2021-06-09:
Hello!
It is a great FC, but I am using copter, and the latest stable firmware is 4.07.
Should I wait or use beta as soon as it is available?
Thank you again for your help!
Rgds,
Andre

Hwurzburg, 2021-06-09:
4.1 beta for copter is already released

AndreHenke, 2021-06-10:
Thanks for all the tips Henry!
I uploaded the beta version and it did not show MSP options. However, when I uploaded the Fmuv3 version, it showed up!
I will make the tests and let you know.
Thanks again!