ArduPilot/ardupilot_wiki #3600: Firmware Limitations shoudl be corrected to say optical flow *is* available on 1MB boards

rmackay9, 2021-06-10:
The firmware limitations page says that optical flow isn't available on 1MB boards but it is available for the Pixhawk-1M at least.
https://ardupilot.org/copter/docs/common-limited-firmware.html#common-limited-firmware

this is in response to this discussion (see discussion with Derek): https://discuss.ardupilot.org/t/copter-4-1-0-beta3-has-been-release-for-beta-testing/71489/41

Hwurzburg, 2021-06-10:
will change, but I need some help on these evolving firmware limitations....I have been trying to keep up, marking them as wiki needed, but they are happening pretty fast and I have probably missed some to tag and no one tags them when they do them...the firmware limitations page is now out of date and I havent had a chance to sort thru all the changes (this weekend hopefully) but will no doubt miss some...I really need those DEVs that merge changes with Wiki impacts to put labels on them since I cant monitor every merge, although I try(@tridge for info)

Hwurzburg, 2021-06-14:
need a little help here.....the Pixhawk-1M boards includes optical flow module but not flowhold mode....what else is optical flow used for???

rmackay9, 2021-06-14:
@Hwurzburg,

Optical flow can be used for Non-GPS navigation so all modes like Loiter, RTL, Auto will all work.  FlowHold's advantage is that it doesn't require a lidar but in practice it doesn't work very well at least according to user support calls.

Hwurzburg, 2021-08-09:
done

closed 2021-08-09 by Hwurzburg