ArduPilot/ardupilot_wiki #3601: firmware upgrade process

tridge, 2021-06-10:
users often make the assumption that to update to a new version they first load the new firmware then load the old parameters. We need to make it clear that the process for upgrading is:
 - load old firmware with old parameters
 - load new firmware, the params will auto-update if needed

Hwurzburg, 2021-06-10:
Why reload existing old firmware? It's already there... Isn't the issue that they just load the new firmware and don't have to reload the old parameters?

Hwurzburg, 2021-06-10:
added note in wiki on firmware load page for existing firmware situations

closed 2021-06-10 by Hwurzburg

rmackay9, 2021-06-11:
Here's another example of this situation: https://discuss.ardupilot.org/t/gps-neo3-suddenly-stopped-working/72310/2

Hwurzburg, 2021-06-11:
I have mistakenly selected the wrong item and had all my params wiped....stupid mistake, but it does happen....