ArduPilot/ardupilot_wiki #3608: Firmware Limitations should say FFT is not available on 1MB boards

rmackay9, 2021-06-16:
This comes from [this discussion in the Copter-4.1 beta testing category](https://discuss.ardupilot.org/t/mot-thst-expo-is-reserved/72463/13)in which a user tried to setup FFT on a Pixhawk1MB.

rishabsingh3003, 2021-06-16:
We also need to update the Avoidance limitations now

Hwurzburg, 2021-06-16:
> This comes from [this discussion in the Copter-4.1 beta testing category](https://discuss.ardupilot.org/t/mot-thst-expo-is-reserved/72463/13)in which a user tried to setup FFT on a Pixhawk1MB.

Actually it's not a 1MB thing..it's HAL_MINIMZE thing and it's already listed as no DSP...but can change that to INFlight FFT...

rmackay9, 2021-06-16:
I wonder why we have both 1MB limits and HAL_MINIMIZE things..

Hwurzburg, 2021-06-17:
thats why it will take me a day or so to sort it all out into the wiki....I would think we could do away with HAL_MINIMIZE and move the non-1MB exclusions into individual defines in each of the boards...KakuteF7,MAtekF405-CAN,MazzyStarDrone,OmnibusF7V2, Pixahaak-1M and 1M-bdshot, fmuv2, and sparky2....in fact, only the F7s really need more than what the F405-Wing has as additional excludes in the HWDEF, correct (in addition to the base 1MB excludes)

Hwurzburg, 2021-08-09:
at the point of this issue, I had subsequently updated the firmware limits page....there has been continuing additional exclusions and I am tracking those to update the wiki as they occur...some still outstanding but are on another issue list, so closing this now

closed 2021-08-09 by Hwurzburg