ArduPilot/ardupilot_wiki #3625: Setting up the Build Environment on Windows10 using WSL1 or WSL2

patrickpoirier51, 2021-06-24:
[Setting up the Build Environment on Windows10 using WSL1 or WSL2](https://ardupilot.org/dev/docs/building-setup-windows10.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Just loaded WSL2 - UBUNTU 20
lsb_release -a

Distributor ID: Ubuntu
Description:    Ubuntu 20.04.2 LTS
Release:        20.04
Codename:       focal

In order to get the connection to the Xserver ,  I had to add this to .bashrc

export DISPLAY=127.0.0.1:0

hendjoshsr71, 2021-06-24:
Odd.

Are you on a Windows Insiders build? I wonder if the new GUI update means you can now use the normal loopback address? Though you wouldn't need xsrvr then.

For my WSL2 also Ubuntu 20.04.02: I must use `export DISPLAY=$(grep -m 1 nameserver /etc/resolv.conf | awk '{print $2}'):0`
which gives the following address. 

veritas@DESKTOP-V9LAGA3:~/ardupilot$ echo $DISPLAY
172.29.208.1:0

patrickpoirier51, 2021-06-25:
Hello
here's my info
OS Name:                   Microsoft Windows 10 Home
OS Version:                10.0.19043 N/A Build 19043
OS Manufacturer:           Microsoft Corporation
OS Configuration:          Standalone Workstation
OS Build Type:             Multiprocessor Free

I think it is more logical to map to loopback address , just like the rest of SITL

Still, this mapping as to be part of the installation process and it is lacking ATM.

Hwurzburg, 2021-11-13:
@hendjoshsr71 @patrickpoirier51  cant do anything on this until approach is resolved....and its a WSL2 only issue, correct?

hendjoshsr71, 2021-11-16:
@Hwurzburg 

Let me ask a few new developers about their setups on WSL2. 

The problem I have with adding this is that changing to  `export DISPLAY=127.0.0.1:0` doesn't work on my WSL2. If I do this on WSL2 then I will get this error in WSL2 when running SITL. 

[image]

My WSL2 requires this: `export DISPLAY=$(grep -m 1 nameserver /etc/resolv.conf | awk '{print $2}'):0`

But there could be reasons I don't know why Patrick's setup requires this change.

Divyateja04, 2021-11-16:
export DISPLAY=$(awk '/nameserver / {print $2; exit}' /etc/resolv.conf 2>/dev/null):0
export LIBGL_ALWAYS_INDIRECT=1

I had to use this, the one in the wiki sometimes says connection refused sleeping, so added the second line, I think the first lines in mine and the above setup are same

hendjoshsr71, 2021-11-16:
> ```
> export DISPLAY=$(awk '/nameserver / {print $2; exit}' /etc/resolv.conf 2>/dev/null):0
> export LIBGL_ALWAYS_INDIRECT=1
> ```
> 
> I had to use this, the one in the wiki sometimes says connection refused sleeping, so added the second line, I think the first lines in mine and the above setup are same

Are you using Ubuntu or Debian? I forget?

Divyateja04, 2021-11-16:
Ubuntu only

> Are you using Ubuntu or Debian? I forget?

Ubuntu Only

hendjoshsr71, 2021-11-17:
I've heard from a few devs and they all have `export DISPLAY=$(grep -m 1 nameserver /etc/resolv.conf | awk '{print $2}'):0` working for them. 

The only one that didn't use that line is using Windows 11 and wslg that doesn't require Xwindows. So I need to add a note for that.

@patrickpoirier51 Maybe we should discuss on Discord and try to find the difference in your system setup to mine? 
I really want our process for new developers to be easy.

patrickpoirier51, 2021-11-18:
@hendjoshsr71 I had the issue on one particular development PC. It may have been a screen driver issue, I dont plan to push investigation deeper. Thanks for following up

closed 2021-11-18 by patrickpoirier51