ArduPilot/ardupilot_wiki #3629: VS Code Debugging: add how to debug startup tasks

hendjoshsr71, 2021-06-27:
Here: https://ardupilot.org/dev/docs/debugging-with-gdb-on-linux.html#vs-code

We should add how to "Launch" debugging in VSCode so users can debug object construction, SITL parameters. 

We may also want to create a new header in the Debugging tree that lists "Debugging with an IDE"

Hwurzburg, 2021-11-13:
@hendjoshsr71 Josh, I know nothing on this topic....someone else will have to write a rough draft and I can polish, if needed....

khancyr, 2021-11-13:
Simple trick to debug initial sequence : put your breakpoint in HAL constuctor or any init phase you want, and reboot !

hendjoshsr71, 2021-11-16:
> Simple trick to debug initial sequence : put your breakpoint in HAL constuctor or any init phase you want, and reboot !

@khancyr Oh nice! That is a lot easier then the launch option for my normal usage. Definitely needs to get added to the wiki.

@Hwurzburg  But I will also include a launch option, autotest, unit test, and more profiles in a PR. 
How would you like to structure this? I was thinking about adding a page under Debugging with GDB -> Debugging SITL Using VSCode

Hwurzburg, 2021-11-16:
@hendjoshsr71 a new page sounds like the best way

PR #3909 (2021-11-17): Dev: add VSCode GDB Debugging Profiles
Closes #3629

Adds VSCode GDB Debugging Profile Information for autotests, attaching, constructor debugging, and unit tests

@Hwurzburg  Here you go. It probably needs a little help in terms of formatting and wording but it should be most of the way there. 

I still need to figure out how to add the JSON file I placed on this page into the actual ArduPilot repo.
closed 2021-11-17 by Hwurzburg