ArduPilot/ardupilot_wiki #3641: Discrepancy in NKF4 documentation

jeffmelville, 2021-07-07:
On https://ardupilot.org/dev/docs/ekf2-estimation-system.html  NK4.{SV,SP, SH, SM} are defined as the innovation test statistics for their corresponding measurements.

On https://ardupilot.org/copter/docs/logmessages.html#nkf4-ekf2-variances, they are defined as the square roots of the variances.

rmackay9, 2021-07-07:
This is probably a question for @priseborough

Hwurzburg, 2021-09-01:
@priseborough Paul?

priseborough, 2021-09-06:
The SV, SP, SH and SM are probably best described as 'Squared Innovation Test Ratios' where values <1 tells us the measurement was accepted and >1 tells us it was rejected. They represent the  square of the (innovation / maximum allowed innovation) where the innovation is the difference between predicted and measured value and the maximum allowed innovation is determined from the uncertainty of the measurement, uncertainty of the prediction  and scaled using the number of standard deviations set by the innovation gate parameter for that measurement, eg EK2_MAG_I_GATE, EK2_HGT_I_GATE, etc

Hwurzburg, 2021-11-13:
@jeffmelville is this now resolved? can we close?