ArduPilot/ardupilot_wiki #3674: How to advise user when DMA issue arises in CRSF

Hwurzburg, 2021-07-28:
CRSF requires DMA on the UART for full performance (anecdotally some features such as param editor may not function also)
what do we advise a user when they get CRSF: serial port has no DMA ?
I recently found one autopilot that had to have its hwdef modified to allow ANY Uart to have full DMA for CRSF.

andyp1per, 2021-07-29:
You know what, it would be pretty easy to extend the warning to tell you which serial port has full DMA

Hwurzburg, 2021-07-29:
That would be a good solution

tridge, 2021-08-02:
note that on H7 we shouldn't need DMA even at high baudrates

Hwurzburg, 2021-08-17:
we still need to decide on what documentation is needed

Hwurzburg, 2021-09-19:
@andyp1per Andy, still getting questions on what to do when CRSF: running on non-DMA serial port  message appears

tridge, 2021-09-20:
I think we should detect the lost packets (presumably CRC errors) and give an arming check failure if packet loss over a threshold
some boards (even non-H7) will handle 400 kBaud with low levels of pkt loss. At 1% loss it would be perfectly usable
@andyp1per would that work?

Hwurzburg, 2026-10-06:
34559 closes this

closed 2026-10-06 by Hwurzburg