ArduPilot/ardupilot_wiki #3676: Option of “HC-SR04 sonar” cannot be found in the configuration menu of RNGFND1_TYPE

BUAAxiaobaige, 2021-07-30:
Hello,

I want to use HC-SR04 sonar for indoor flying, but when I tried to configure it referring the guide of [this page](https://ardupilot.org/copter/docs/common-rangefinder-hcsr04.html), I cannot find the option of “HC-SR04 sonar” in the menu of RNGFND1_TYPE(neither in QGroundControl nor Mission Planner).

I have tried several versions of ardupilot firmware but none of them works.

Can anyone help me? Thanks a lot in advance.

hendjoshsr71, 2021-07-30:
Hi @BUAAxiaobaige ,

Please post your hardware type, firmware version and Mission Planner version.

I found it in Mission Planner using DEV firmware. I have feeling it is not in your firmware as it first appeared in 4.1.0-beta3 it looks like. If you are comfortable using beta firmware you could use the latest 4.1.0-beta

See below.

[image]

Hwurzburg, 2021-07-30:
its been in the code since June of last year

BUAAxiaobaige, 2021-07-31:
> Hi @BUAAxiaobaige ,
>
> Please post your hardware type, firmware version and Mission Planner version.
>
> I found it in Mission Planner using DEV firmware. I have feeling it is not in your firmware as it [...]

oh it works, I kept trying firmware of versions lower than 4.1 and now change it to 4.1.0 and I can find the option of "sr04"

thanks a lot !

Regard.

closed 2021-07-31 by BUAAxiaobaige