ArduPilot/ardupilot_wiki #3688: 4.2 Wiki tasks (should be in upcoming features)

Hwurzburg, 2021-08-07:
- [ ] 17539 New GUIDED mode position controller(lthall)
- [ ] 17613 Ride Along JSON (iampete)
- [ ] 17651 ONIF camera in SITL/Linux
- [ ] 17668 Script backend for interpolated matrix (iampete)
- [ ] 176978 Land reposition in Q_OPTIONS (tridge)
- [ ] 17790 Airspeed disabled AUX switch (pbarker)
- [ ] 17894 FRSKY_OPTIONS parameter (yaapu)
- [ ] 17945 Matek F765-SE board(tridge)
- [ ] 17975 --udp option for sim_vehicle (tridge)
- [x] 17979 Torqeedo motor (rmakay9)
- [ ] 17986 CHUTE_OPTIONS (tridge)
- [ ] 17990 CRSF v3 (andypiper)
- [ ] 17992 USB passthru
- [ ] 17998 MAMBA Basic MK3.5 board (hw)
- [ ] 18018 PWM_BRD_COUNT removal (tridge)
- [ ] 18043 EXPO param for Plane manual/acrp/trainer modes
- [ ] 18133 switch to gcc 10.2.1 (tridge)
- [ ] 18137 LOG_FILE_RATEMAX with note on DSF.Dp (tridge)
- [ ] 18184 ROMFS_custom integration into ROMFS (tridge)

rmackay9, 2021-08-08:
"New GUIDED mode position controller" is included in 4.1 actually.  we should update this page https://ardupilot.org/dev/docs/mavlink-commands.html

HefnySco, 2021-09-06:
"18337 Webots2012b compatibility (@HefnySco)"
yes. Webots2021b is replacement of old one and the old files are not compatible with it.

drtrigon, 2021-11-30:
18929 Cam control Lua applet: pull request was accepted. Is there any other (administrative) thing to do? @IamPete1 can you instruct me what else needs to be done?

bnsgeyer, 2022-02-01:
@Hwurzburg Tradheli autotune (#14619) is expected to be merged for 4.2.  I am writing the wiki and it will be ready to merge directly.  I will work on the other two after I get autotune done