ArduPilot/ardupilot_wiki #3690: Dev: SITL: gimbals instructions completely out of date

IamPete1, 2021-08-08:
Caused https://github.com/ArduPilot/ardupilot/issues/15179

https://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html#adding-a-gimbal

Is just completely wrong, we don't even have those params anymore.

It might be better to just delete them until someone get to re-wring....

Hwurzburg, 2021-08-09:
@IamPete1 Pete I am kinda at a loss on this....I can run SITL fine with the -M option...all the parameters mentioned in the section for mnt_type and servo functions are valid....so dont quite understand the issue here...

IamPete1, 2021-08-09:
It looks to me that `-M` simulates a mavlink gimbals not a servo one. At the least we should update `RC6_FUNCTION` to `SERVO6_FUNCTION`. I should add that I have not actually tested the correct way to set this up.

Hwurzburg, 2021-08-10:
fixed.....was just looking at function numbers when I checked....if you find something wrong with this in SITL (I dont know how to test other than it runs) or how to setup a servo gimbal in SITL, open a new issue please...thanks

closed 2021-08-10 by Hwurzburg