ArduPilot/ardupilot_wiki #3695: no drone appears in gazebo

skennetho, 2021-08-10:
Hi. im trying to do a SITL in my laptop (ubuntu20.04, gazebo11)
i've done the procedures in the guide "Using Gazebo Simulator with SITL" (https://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html)

but when i do
$> ~/ardupilot_gazebo$ gazebo --verbose worlds/iris_arducopter_runway.world 
gazebo comesout well but without drone showing...

i tried 
$> echo 'export GAZEBO_MODEL_PATH=~/ardupilot_gazebo/models' >> ~/.bashrc
$> source ~/.bashrc
$> source /usr/share/gazebo/setup.sh
$> source /usr/share/gazebo-11/setup.sh
still doesn't work.

**this is what i got in my terminal:**
~/ardupilot_gazebo$ gazebo --verbose worlds/iris_arducopter_runway.world 
Gazebo multi-robot simulator, version 11.7.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.7.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:4560
[Msg] Publicized address: 172.16.12.51
[Msg] Connected to gazebo master @ http://127.0.0.1:4560
[Msg] Publicized address: 172.16.12.51
Error [parser.cc:525] parse as sdf version 1.7 failed, should try to parse as old deprecated format
Error Code 4 Msg: Required attribute[xmlns:xacro] in element[sdf] is not specified in SDF.
Error Code 8 Msg: Unable to parse sdf element[sdf]
[Msg] Loading world file [/home/awesometech/ardupilot_gazebo/worlds/iris_arducopter_runway.world]

skennetho, 2021-08-10:
solved it by my own.
the problem was in 'models/iris_with_ardupilot/model.sdf'.

in line 2, it was...
<sdf version="1.6" xmlns:xacro='http://ros.org/wiki/xacro'>
and i change it into...
<sdf version="1.6" >

now when i run...
$> gazebo --verbose worlds/iris_arducopter_runway.world 
iris drone appears in gazebo!

very little thing but hope this will help some beginners like me :smile:

closed 2021-08-10 by skennetho

MOVAX19, 2021-09-21:
Hi, @skennetho.  I'm having the same issue as this; I tried to follow the solution that you proposed but it seems that it is incomplete. Could you share it again, please?

skennetho, 2021-09-22:
> Hi, @skennetho.  I'm having the same issue as this; I tried to follow the solution that you proposed but it seems that it is incomplete. Could you share it again, please?

Oops. i editted the solution! Now you can see the lines.
Hope it could help!

Davysseus, 2021-09-28:
> solved it by my own. the problem was in 'models/iris_with_ardupilot/model.sdf'.
>
> in line 2, it was...
>
> ```
> <sdf version="1.6" xmlns:xacro='http://ros.org/wiki/xacro'>
> ```
>
> and i change it into... [...]

I did this and it fixed it. Thanks for sharing.

How did you figure it out though? I don't actually follow what is happening.