ArduPilot/ardupilot_wiki #3753: Complete Parameter List

mattdholmes00, 2021-09-07:
[Complete Parameter List](https://ardupilot.org/copter/docs/parameters.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.
On the AHRS_ORIENTATION description, is the second action relative to the original orientation, or relative to the new board orientation after the first action is completed?  I.e. after I for example pitch 180 and then yaw 90, does the FC end up facing to the left or to the right?  If the yaw 90 is relative to the original coordinate system, it ends up facing left; if the yaw 90 is relative to the new board coordinate system, it ends up facing right.  Whichever it is, this should be specified in the description.

Hwurzburg, 2021-09-29:
@tridge bump on answer needed

TunaLobster, 2021-09-29:
The rotations should be applied in Euler angle order. For an aircraft that would be yaw, pitch, roll in that order. I've only used single axis rotations with those parameters though.