ArduPilot/ardupilot_wiki #3776: need to update dev wiki for vscode

tridge, 2021-09-22:
The eclipse docs are ancient, and all oriented around px4 builds. That should be fixed or removed
then vscode docs, which I'm told is what all the cool kids use these days, should be updated to the level we had for eclipse
buzz volunteered to help!

hendjoshsr71, 2021-09-22:
Related issue here about adding "launch" debug tasks in VSCode. https://github.com/ArduPilot/ardupilot_wiki/issues/3629

tridge, 2021-09-23:
there is a better eclipse page here:
https://ardupilot.org/dev/docs/building-setup-windows-eclipse.html#building-setup-windows-eclipse
maybe we just remove the px4 oriented eclipse page?

tridge, 2021-09-23:
there is some info on vscode here:
https://ardupilot.org/dev/docs/debugging-with-gdb-on-linux.html?highlight=vscode

rmackay9, 2021-09-23:
Yes, I think we should

- archive the old eclipse page: https://ardupilot.org/dev/docs/editing-the-code-with-eclipse.html
- modify the landing page (https://ardupilot.org/dev/docs/code-editing-tools-and-ides.html) so the Eclipse link points to the new page (https://ardupilot.org/dev/docs/building-setup-windows-eclipse.html)
- on the new eclipse page (https://ardupilot.org/dev/docs/building-setup-windows-eclipse.html) remove:
  - the line, "These instructions replace the older method which relied on Make."
  - the warning, "ArduPilot no longer supports make.."
  - the section, "Remove Pixhawk Toolchain"

hendjoshsr71, 2021-11-17:
PR #3909 completes the debugging part.

Other thoughts I had were to include a list of useful settings in this proposed page.
Probably most important being setting up User Setting Sync to the cloud.

TunaLobster, 2022-07-08:
I think this is mostly complete now. Any comments on what might still be missing?

Hwurzburg, 2022-07-11:
no need to discuss the two PRs...one changing away from Eclipse and the VScode one that was pacing it...will mark for the call tonight