ArduPilot/ardupilot_wiki #3887: Complete Parameter List

ardugitter, 2021-11-07:
[Complete Parameter List](https://ardupilot.org/copter/docs/parameters.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.
 
The DEV_OPTIONS entry is missing bit 2 which is meant to be used to control MAVLINK_MSG_ID_SET_ATTITUDE_TARGET thrust field interpretation as here:
https://github.com/ArduPilot/ardupilot/pull/14918

When bit 2 is unset, thrust is interpreted in the same way as altitude hold mode, 1 causing the drone to move upwards and 0 causing it to move downwards at stable velocities as dictated by WPNAV_SPEED_*.
When bit 2 is set, a thrust of 1 is interpreted to mean apply maximum power to motors and a thrust of 0 means zero power applied to motors.

Hwurzburg, 2021-11-08:
this really is not something that can be changed in the wiki...has to be done in the ArduPilot code itself....but I will look into doing a PR for that...

rmackay9, 2021-11-08:
Actually I think we put this bit into GUID_OPTIONS. https://github.com/ArduPilot/ardupilot/blob/Copter-4.1/ArduCopter/Parameters.cpp#L1020

It was temporarily in DEV_OPTIONS during the development period (when the firmware string was Copter-4.1.0-DEV).

I think I'll close this issue if that's OK but I'll monitor for responses.

closed 2021-11-08 by rmackay9

ardugitter, 2021-11-08:
Sorry about that. Thanks for clearing up my confusion. I'll check the current parameter list next time.