ArduPilot/ardupilot_wiki #3928: Why /ArduCopter/ArduCopter.elf file doesn't exist?

cnpcshangbo, 2021-11-22:
Hello,
I am trying to follow this documentation
https://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html#sitl-without-mavproxy-tcp
to run the "elf" file. However, this file doesn't exist in the ArduCopter folder. 
```
bo@bo-Precision-3650-Tower:~/code/ardupilot/ArduCopter$ ls | grep elf           
bo@bo-Precision-3650-Tower:~/code/ardupilot/ArduCopter$ 
```
I was able to run the SITL using the python script. Are there some steps I need to take to generate the elf file or that file is in a different location?
Thanks,
Bo

khancyr, 2021-11-22:
Hello,

Indeed, the wiki Is not clear on this, the binaries are in build/sitl/bin

Hwurzburg, 2021-12-02:
@pierre I do not get an .elf file in the [image]

khancyr, 2021-12-02:
@Hwurzburg yep, we don't have the .elf and the wiki is also wrong at mentioning that the binary will by in vehicle directory like ./ArduCopter !

hendjoshsr71, 2021-12-02:
Just for my clarity the only difference between the .elf and the binaries as they are normally is changing the extension to `.elf``?

We do that in the cygwin builds for mission planner. We also use the elf_diff tool on the files as is without the extension in out github actions.

Hwurzburg, 2021-12-02:
those shown in that Linux directory are the executables I think even without the .elf extension...correct?

hendjoshsr71, 2021-12-02:
> those shown in that Linux directory are the executables I think even without the .elf extension...correct?

Yes definitely. You can even run them directly. I'm not sure how well doing that is documented on the wiki.