ArduPilot/ardupilot_wiki #3934: Copter: EKF3 Body Drag Coefficient Estimation & Baro Compensation

hendjoshsr71, 2021-11-27:
Request:  for how to use  the EKF3 Body Drag Coefficient Estimation  feature as noted on this [Discuss Post](https://discuss.ardupilot.org/t/wind-speed-estimation-and-barometer-interference-compensation/78718/8)

Link to Paul's Presentation: [here ](https://www.youtube.com/watch?v=xVVtvVuZGQE)
Relevant PR that brought this feature [here](https://github.com/ArduPilot/ardupilot/pull/15774)

Hwurzburg, 2021-11-27:
Already in the wiki under advanced configuration

closed 2021-11-27 by Hwurzburg

hendjoshsr71, 2021-11-28:
Thanks for pointing that out! I think I only searched for "wind" and the EKF parameter names and didn't come up with much.

I do think we should get page eventually that walks through this procedure in more than video form. But I realize this is a quite an advanced feature so not for the top of the list.