ArduPilot/ardupilot_wiki #4090: Modules

AndKe, 2022-01-31:
[Modules](https://ardupilot.org/mavproxy/docs/modules/index.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

How can I add help/description pages for two more modules?

Hwurzburg, 2022-02-03:
depends on if you are comfortable setting up a local wiki repo and build environment or not....if so:
1. add a new .rst page for each module description in the MAVProxy/source/docs/modules sub-dir
2. add to TOC list on the above page
submit a PR to the wiki

if not....create the text for the above two pages and I can format and add as a PR

AndKe, 2022-02-10:
@Hwurzburg 
I've made those plugins years ago, and just hope that more can find them useful if they know about them :)
Please see two documents I've made:
https://docs.google.com/document/d/1Io6bautqhZ9JzUWK2rF8EnRaGxDIVNUqMl0OZU6sgFA/edit?usp=sharing
https://docs.google.com/document/d/1hEpoqQ8qjFPc2i7QYa40K5UBK_3CgQYPUidNc5EFJlU/edit?usp=sharing

Feel free to change anything as you see fit, English is not my native language.
Thank you.

Hwurzburg, 2022-02-10:
I will work on the moving home doc....but I dont see the park module in MAVProxy

AndKe, 2022-02-10:
wow.. good catch! 
please see https://github.com/ArduPilot/MAVProxy/pull/1009

Hwurzburg, 2022-02-10:
just tagged it for devcall

AndKe, 2022-02-10:
Thank you, I have no idea why it's failing CI, the closest thing I understand from that is that have something to do with windows:

```
ISCC.exe: D:\a\_temp\9dc7b4c4-a4d1-49[13](https://github.com/ArduPilot/MAVProxy/runs/5141917519?check_suite_focus=true#step:9:13)-b0[14](https://github.com/ArduPilot/MAVProxy/runs/5141917519?check_suite_focus=true#step:9:14)-e7020d32b664.ps1:5
Line |
   5 |  ISCC.exe /dMyAppVersion=$env:VERSION mavproxy.iss
     |  ~~~~~~~~
     | The term 'ISCC.exe' is not recognized as a name of a cmdlet, function, script file, or executable
     | program. Check the spelling of the name, or if a path was included, verify that the path is correct
     | and try again.

Error: Process completed with exit code 1.
```

none of that is related to what I did in the module.  I hope that whoever reviews the error understands that? ...or maybe people will just ignore it, thinking there is something very wrong with it?

AndKe, 2022-03-25:
@Hwurzburg   now it's failing in some unrelated part of the windows build:
```
Downloading pyinstaller-4.3.tar.gz (3.7 MB)
     ---------------------------------------- 3.7/3.7 MB 33.4 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error
  
  Preparing metadata (pyproject.toml) did not run successfully.
  exit code: 1
```

AndKe, 2022-04-22:
@Hwurzburg  - the Park module is now in master :)   - please add the docs for it.
https://github.com/ArduPilot/MAVProxy/pull/1009

Hwurzburg, 2022-04-25:
> @Hwurzburg - the Park module is now in master :) - please add the docs for it. [ArduPilot/MAVProxy#1009](https://github.com/ArduPilot/MAVProxy/pull/1009)

so this is just a notification on the GCS....I see that you can change the radius from 2M tolerance...for a loitering plane it would be a bit bigger than the loiter radius, correct?

AndKe, 2022-04-26:
@Hwurzburg Correct, for a plane, one would increase radius, and set the "park on" when the plane is on the point it will circle around.
As for movinghome; you can certainly use uBlox ..any ublox GNSS reciever can output NMEA, but the plugin reads only NMEA 0183 (not UBX protocol)  - nor does it recognize ublox devices, and configure them to output NMEA.
If you wish to use a uBlox device, then it's really easy - use u-Center (ublox software - can be run in wine on non-windows too)  - anable NMEA output alongside UBX, and store config.

Hwurzburg, 2022-05-21:
@AndKe please review https://github.com/ArduPilot/ardupilot_wiki/pull/4354

closed 2022-05-21 by Hwurzburg