ArduPilot/ardupilot_wiki #4099: Iflight beast h7 page - possible useful information to add

payor-ma, 2022-02-04:
Hey not sure if this is worth putting up on the wiki.  

On this page https://github.com/ArduPilot/ardupilot_wiki/blob/master/common/source/docs/common-iflight-beasth7AIO.rst
It is stated that the iflight beast h7 board does not support i2c for a compass.  Whilst this is correct, I found it to be a desirable target and figured out a relatively accessible way to break out i2c for compass and other sensor support.  Unsure whether such a modification belongs on the wiki but maybe others may find it useful.

[image]

Next to the barometer, which is hooked up via i2c there are two resistors which are 2.2k pull ups for the SDA and SCL.  tacking wires onto the right side of these resistors based on the orientation in the diagram, allows expansion of the i2c bus.  Here's evidence:

[image]
[image]
[image]
[image]

Availability of i2c for compass and other sensors may be of high value to other people looking to use this board with autopilot.

payor-ma, 2022-02-04:
I'm happy to produce some clear documentation and put in a pull request but I'm unsure if this kindof stuff belongs on the wiki.

Hwurzburg, 2022-02-04:
yes,we do put this kind of into in....see  https://ardupilot.org/plane/docs/common-furiousfpv-f35.html
however, since this is directly to components rather than solder pads I would have warnings about potential circuit damage to the components...ie requires skilled soldering techniques....

payor-ma, 2022-02-27:
I'll close this issue and just put in a pull request when I have some time to do a write up.

closed 2022-02-27 by payor-ma