ArduPilot/ardupilot_wiki #4155: Using Gazebo Simulator with SITL

furkanozdemi, 2022-02-24:
I have a problem.

furkanozdemi, 2022-02-24:
python sim_vehicle.py -v ArduCopter
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/furkan/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Setting top to                           : /home/furkan/ardupilot 
Setting out to                           : /home/furkan/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Using toolchain                          : native 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
CXX Compiler                             : g++ 9.3.0 
Checking for need to link with librt     : not necessary 
Checking for feenableexcept              : yes 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for HAVE_MEMRCHR                      : yes 
Configured VSCode Intellisense:                : no 
Checking for program 'python'                  : /usr/local/bin/python 
Checking for python version >= 2.7.0           : 2.7.0 
Checking for program 'python'                  : /usr/local/bin/python 
Checking for python version >= 2.7.0           : 2.7.0 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Scripting                                      : enabled 
Scripting runtime checks                       : enabled 
Debug build                                    : disabled 
Coverage build                                 : disabled 
SITL 32-bit build                              : disabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (1.057s)
SIM_VEHICLE: Building
SIM_VEHICLE: "/home/furkan/ardupilot/modules/waf/waf-light" "build" "--target" "bin/arducopter"
Waf: Entering directory `/home/furkan/ardupilot/build/sitl'
Traceback (most recent call last):
  File "/home/furkan/ardupilot/modules/waf/waflib/Scripting.py", line 158, in waf_entry_point
    run_commands()
  File "/home/furkan/ardupilot/modules/waf/waflib/Scripting.py", line 251, in run_commands
    ctx = run_command(cmd_name)
  File "/home/furkan/ardupilot/modules/waf/waflib/Scripting.py", line 235, in run_command
    ctx.execute()
  File "Tools/ardupilotwaf/ardupilotwaf.py", line 202, in execute
    return execute_method(self)
  File "/home/furkan/ardupilot/modules/waf/waflib/Scripting.py", line 573, in execute
    return execute_method(self)
  File "/home/furkan/ardupilot/modules/waf/waflib/Build.py", line 231, in execute
    self.execute_build()
  File "/home/furkan/ardupilot/modules/waf/waflib/Build.py", line 244, in execute_build
    self.recurse([self.run_dir])
  File "/home/furkan/ardupilot/modules/waf/waflib/Context.py", line 286, in recurse
    user_function(self)
  File "/home/furkan/ardupilot/wscript", line 765, in build
    _load_pre_build(bld)
  File "/home/furkan/ardupilot/wscript", line 749, in _load_pre_build
    brd.pre_build(bld)
  File "Tools/ardupilotwaf/boards.py", line 415, in pre_build
    self.embed_ROMFS_files(bld)
  File "Tools/ardupilotwaf/boards.py", line 429, in embed_ROMFS_files
    import embed
  File "Tools/ardupilotwaf/embed.py", line 10, in <module>
    import os, sys, tempfile, gzip
  File "/usr/local/lib/python2.7/gzip.py", line 9, in <module>
    import zlib
ImportError: No module named zlib
SIM_VEHICLE: Build failed
SIM_VEHICLE: Killing tasks

hendjoshsr71, 2022-02-24:
Closing this isn't a Wiki Issue. You should open a page on https://discuss.ardupilot.org/c/development-team/5

Your Python Version is 2.7 you should use python3 : Your command `python` is pointing to python2
Did you try installing the missing module `zlib` for python2?

Is this a recent install? If you used our install scripts you should not need to prepend the command with python as the ardupilot folder should be on the PATH.

closed 2022-02-24 by hendjoshsr71