ArduPilot/ardupilot_wiki #4274: 4.3 Wiki (should be in upcoming features)

Hwurzburg, 2022-04-02:


tridge, 2022-04-02:
ship landing would be good to document: https://discuss.ardupilot.org/t/ship-landing-support/82772
it is in 4.2

Hwurzburg, 2022-04-02:
its on the 4.2 list already as undocumented...

andyp1per, 2022-07-17:
@Hwurzburg 

https://github.com/ArduPilot/ardupilot/pull/20296
https://github.com/ArduPilot/ardupilot/pull/20230

rmackay9, 2022-11-20:
@tridge and @Hwurzburg we should add use of external ahrs like the VectorNav products to this list I think.  We have [an issue from a user struggling with the setup here ](https://github.com/ArduPilot/ardupilot/issues/22235)and I can't immediately find any documentation on the setup on our wiki.

rmackay9, 2024-11-24:
@Hwurzburg I've marked the QURT HAL one as done now that we've added the VOXL2 page https://github.com/ArduPilot/ardupilot_wiki/pull/6402

andypiper, 2024-11-26:
@rmackay9 @Hwurzburg please correct the references to my username in this issue to refer to @andyp1per thank you.

Hwurzburg, 2024-11-26:
done... unfortunately these do not generate any notifications...was more for me to know who to bug

andypiper, 2024-11-26:
I think my handle is still in there (under PID notches) and I have to keep unsubscribing from the issue for some reason.

Hwurzburg, 2024-11-26:
missed it,sorry