ArduPilot/ardupilot_wiki #4280: USB connection and upload from WSL2

RuffaloLavoisier, 2022-04-05:
A way to connect USB on WSL2 has been officially developed. This will further enhance the performance of the development environment. I think this part needs to be reflected in the [document](https://github.com/ArduPilot/ardupilot_wiki/blob/master/dev/source/docs/building-setup-windows10.rst#wsl1-vs-wsl2) and updated.

Hwurzburg, 2022-04-05:
its not actually part of official windows but a third party open source project...it does work however....the docs do need updating for  Win11 since it integrates LINUX WSL2 fully except for USB, which needs that add-on...

RuffaloLavoisier, 2022-04-06:
It's not an official method, but I think we need to let others know that we can improve the development environment. how about you?

Hwurzburg, 2022-04-06:
its on my list to add a Win11 page...the comment was that its not an official Microsoft release...third party open source

Hwurzburg, 2024-10-17:
added

closed 2024-10-17 by Hwurzburg