ArduPilot/ardupilot_wiki #4290: Arduplane ACRO mode documentation assumes ACRO_LOCKING is enabled, but it defaults to disabled.

cbf123, 2022-04-13:
The documentation for ACRO mode (https://github.com/ArduPilot/ardupilot_wiki/blob/master/plane/source/docs/acro-mode.rst) is written assuming that ACRO_LOCKING is enabled, but it defaults to disabled.

Here is an example:
When flying in ACRO the aircraft will try to hold its existing attitude if you have no stick input. 

I believe this would only be true if ACRO_LOCKING is enabled.

Hwurzburg, 2022-04-13:
it indeed will TRY to prevent any un-commanded rates leading to an attitude change....does so well in the short term, but imperfection will allow long term changes....ACRO_LOCKING actually trys to stop attitude changes directly feeding error corrections into the rate controllers....like TAIL HOLD in a heli does not actually  prevent heading drift, but "locks" the tail against short term rate errors...heli tail hold would actually need a compass to actually be a heading hold instead of a heading rate change prevention...
I think the note explains it well enough

Hwurzburg, 2022-04-14:
this is how rate controller (ACRO) is generally understood to work...as the note states, ACRO_LOCKING is required to hold a given pitch/roll attitude over the long term and to counteract long term drift...

cbf123, 2022-04-14:
It's worth noting that for ArduCopter, ACRO mode defaults to what ArduPlane mode does with ACRO_LOCKING enabled.   To get the equivalent of ACRO_LOCKING disabled you have to disable air mode via ACRO_OPTIONS.

This is *not* immediately obvious, and the difference should be made clear in the documentation.

I'm relatively new to flight controllers, and had some knowledge of multicopters and of fixed-wing gyros, and it was impossible to reliably tell from the Arduplane docs that ACRO mode provided basic gyro stabilization.  I had to actually test it out to be sure.

Hwurzburg, 2022-04-14:
feel free to submit  a PR for suggested changes to the documentation

FPVZaphod, 2024-09-15:
That's quite old now, but I's like to request the possibility to change this in-flight. Can we have two acro modes, one with disabled locking, and one with enabled? I fly 90% of the time in Acro, and it would be nice to have both available. As an example, with enabled locking for landing, and disabled locking for freestyle / acrobatic flying.
Maybe ACRO and ACRO-L?

Hwurzburg, 2024-09-16:
which autopilot are you using? LUA scripting would avoid flash cost which is at a premium....the basis of this issue has been addressed and I will close tomorrow...you can raise a feature request as a code issue

FPVZaphod, 2024-09-16:
All are F405s, so quite limited (if not impossible) scripting options :-/

Hwurzburg, 2024-09-16:
you can add a feature request in code github

closed 2024-09-16 by Hwurzburg