ArduPilot/ardupilot_wiki #4295: Add X-Plane data in and out settings to sitl-with-xplane.rst

radiohound, 2022-04-15:
There was a step left out of the instructions for sitl-with-xplane.rst . When I used the additional instructions below (from x-plane-hardware-in-the-loop-simulation.rst) sitl with xplane worked. With X-Plane's initial default settings, no data will go in or out of X-Plane.

These are the screen shots that show the necessary settings you need to set in the Settings Menu – Data Input and Output:
[image]
[image]

TunaLobster, 2022-04-18:
I reviewed the X-Plane SITL code and retested this. With ArduPilot running and no X-Plane data being output (none of the boxes checked), ardupilot SITL waits for data to show up. Once the first packet shows up, ArduPilot sends commands to set the rest of the data output settings and then SITL continues on like normal.

I do understand the issue though. There are 2 X-Plane pages with complimentary information. Lets leave this open and I will consolidate the information down and update for X-Plane 11 where needed.

Hwurzburg, 2022-04-22:
closed by above

PR #4311 (2022-04-22): Dev: Update X-Plane docs
A few of the screenshots are older, but you can still make out what is going on between the newer Mission Planner and X-Plane 11 moving things around a little bit. Removed unused screenshot file.

Closes #4295
closed 2022-04-22 by Hwurzburg