ArduPilot/ardupilot_wiki #4301: Get and Set FlightMode - MAVLink doco - small edits

pauljeffress, 2022-04-18:
[Get and Set FlightMode](https://ardupilot.org/dev/docs/mavlink-get-set-flightmode.html) 
In the table for param1 it says "MAV_MODE_FLAG_CUSTOM_MODE_ENABLED=1(", should there be some additional text there after the open bracket?
Also, for param2, the "flightmode number", it would be cool if there were links to what those numbers are/mean, for the different ArduPilot platforms (Rover, Plane etc).
I think this is the one for Plane? https://ardupilot.org/plane/docs/flight-modes.html#
I think this is it for Rover? https://ardupilot.org/rover/docs/rover-control-modes.html
Thanks,
Paul

pauljeffress, 2022-04-18:
I just noticed that while the links I mentioned talk about each platforms modes, they don't actually give you the mode name to mode number mapping that you need when using the MAVLink MAV_CMD_DO_SET_MODE command as referenced on 
[Get and Set FlightMode](https://ardupilot.org/dev/docs/mavlink-get-set-flightmode.html)
What we really need is the info that I can only find in the code itself like for Rover
https://github.com/ardupilot/ardupilot/blob/master/Rover/mode.h#L21
where it shows the relevant enum.
`    // Auto Pilot modes
    // ----------------
    enum Number : uint8_t {
        MANUAL       = 0,
        ACRO         = 1,
        STEERING     = 3,
        HOLD         = 4,
        LOITER       = 5,
        FOLLOW       = 6,
        SIMPLE       = 7,
        AUTO         = 10,
        RTL          = 11,
        SMART_RTL    = 12,
        GUIDED       = 15,
        INITIALISING = 16
    };`

rmackay9, 2022-04-18:
@pauljeffress,

Txs for the report.  I'll remove the stray "(".

Near the top of the page it has links, "see [FLTMODE1](https://ardupilot.org/copter/docs/parameters.html#fltmode1) for flightmode numbers" but maybe we can repeat the link down in the param2 field.

pauljeffress, 2022-04-18:
@rmackay9 ,
Can't believe I didn't see the links at the top of the page..... :( opps.
That said, given this page is all about using MAVLink to set the Flight Mode, I think it might be good to also point the reader to the related MAVLink protocol dialect for ArduPilot, that's taught me a lot.  The Flightmodes are defined in there in ardupilotmega.h within the MAVLink library....I'll find a GitHub link in a sec.
Tha ks again for all the great work you do here @rmackay9

pauljeffress, 2022-04-18:
For Plane it's https://github.com/mavlink/c_library_v1/blob/079d909ee331205d47652b208057d265338dc839/ardupilotmega/ardupilotmega.h#L821
For Rover it's https://github.com/mavlink/c_library_v1/blob/079d909ee331205d47652b208057d265338dc839/ardupilotmega/ardupilotmega.h#L906
Now I  don't know enough to know if those links are the right way to point to the enums in Github, but I really like going to the definitive source.  What do you think?

rmackay9, 2022-04-20:
No troubles, thanks for all this.

I've added a link to the mavlink repo although I fear it will become out of date quickly.. still, it might help anyway.

rmackay9, 2022-04-21:
I think we can close this now..

closed 2022-04-21 by rmackay9