ArduPilot/ardupilot_wiki #4329: Release Notes

AndKe, 2022-05-01:
[Release Notes](https://ardupilot.org/plane/docs/project-news.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

this page gives an incorrect impression of the Arduplane to have ended development years ago.  - if nothing else, it should point to the [github releaselog](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/ReleaseNotes.txt)     - possibly the same issue with copter/other pages.

Hwurzburg, 2022-05-01:
It already does...first line

AndKe, 2022-05-01:
You are absolutely right, that requires the user to "discover" that the first one is a link, not just another heading.
Maybe is "Current Releases" said "Link to Current Releases"   - that would be obvious - or if there was a link under it (it looks like a heading, just as "Older Releases")    - on the other hand, maybe it's just me :)  You decide :)

Hwurzburg, 2022-05-02:
done...

closed 2022-05-02 by Hwurzburg