ArduPilot/ardupilot_wiki #4336: Kakute H7 mini wiki update on battery monitoring

HuguesD, 2022-05-09:
The wiki for the Kakute H7 mini should be updated on the battery configuration section to indicate that if the H7 mini is used with its 4-in-1 Tekko ESC, then

-ESC telemetry is usable via the built-in Serial 1 but does not provide global battery monitoring for current and voltage measurement (only individual ESC measurements can be viewed in for example the status window on mission planner)
-Instead to measure global voltage and current the following settings must be configured to measure voltage and current via the inputs of the H7 mini flight controller. The correct battery setting parameters are:

[BATT_MONITOR](https://ardupilot.org/copter/docs/parameters.html#batt-monitor) 4 when using a Kakute 4-in-1 ESC sold in the H7 mini stack (Tekko 32 )
[BATT_VOLT_PIN](https://ardupilot.org/copter/docs/parameters.html#batt-volt-pin) 10 (corresponds to the voltage measurement pin on the Kakute H7 mini which connects to the 4-in-1 ESC)
[BATT_CURR_PIN](https://ardupilot.org/copter/docs/parameters.html#batt-curr-pin) 11 (corresponds to the current measurement pin on the Kakute H7 mini which connects to the 4-in-1 ESC)
[BATT_VOLT_MULT](https://ardupilot.org/copter/docs/parameters.html#batt-volt-mult) 11.1
[BATT_AMP_PERVLT](https://ardupilot.org/copter/docs/parameters.html#batt-amp-pervlt) 59.5 when using the Kakute 4-in-1 ESC sold in the H7 mini stack (Tekko 32)

Hwurzburg, 2022-05-16:
I dont think all this is needed....perhaps a simple addition to existing "BATT_AMP_PERVLT" line to say that if Tekko32 ESC is used, the value would be 59.5, as an example

Hwurzburg, 2022-05-19:
updated wiki...thanks