ArduPilot/ardupilot_wiki #4338: disable gps during flying and in the specific conditions

John-Mccarty, 2022-05-09:
hi I want to stopping using gps during flying in some conditions : "if sat count under the 10 , if the control mode is auto or rtl , if the plane is arming and if the gps hdop error is too high". I am doing it inside Arduplane.cpp file with force_disable function and it`s stop using gps and make it`s status to no_fix but after that it doens`t getting using gps even in the all condiotions.How can i do stop using gps just in the conditions of above?

rmackay9, 2022-05-10:
Hi @John-Mccarty,

I think you could use a Lua script if the autopilot you're using has a high powered CPU (like an H7 on the CubeOrange, Durandal, etc).

The detection of the GPS state could be done as it is done in this script that I wrote recently. https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/copter-deadreckon-home.lua

Then the [AHRS::set_posvelyaw_source_set()](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/generator/description/bindings.desc#L51) could be used to change the EK3_SRCx_POSXY/VELXY to not use the GPS.

rmackay9, 2022-05-10:
Ah, this issue was raised on the wiki issues list.  This should really be on the main flight code issue list or raised in the forums.

John-Mccarty, 2022-05-13:
Thanks for the response.I tried to change to ekf_source to set_posvelyaw_source to camera it doesnt effect until I disable gps when I disable gps then it automaticly switch the other source(camera).and I am using the latest firmware and t265 camera with vio