ArduPilot/ardupilot_wiki #4376: GPS-for-Yaw: complete instructions for DroneCAN GPSs

rmackay9, 2022-05-31:
[These instructions](https://discuss.ardupilot.org/t/copter-4-2-0-beta1-available-for-beta-testing/82460/16) from @MallikarjunSE explain the parameter changes required for GPS-for-yaw when using DroneCAN GPSs.

[A user said they were useful and required](https://discuss.ardupilot.org/t/gps-yaw-cannot-be-used/83175/8) to get the setup working (apparently our wiki instructions were not sufficient).

The bits we seem to be missing for DroneCAN GPSs are:

- GPS_AUTO_CONFIG= 2 (Enable automatic configuration for DroneCAN)
- GPS_DRV_OPTIONS= 8 (Use dedicated CAN port between GPSs for moving baseline)
- GPS1_CAN_OVRIDE and GPS2_CAN_OVRIDE based on your node. Option is available under CAN GPS Order on MP.
- On the AP-Periph side, make sure define GPS_MOVING_BASELINE 1 is defined

I think maybe we should split the setup instuctions into "Serial GPSs" and "DroneCAN GPSs" and

MallikarjunSE, 2022-05-31:
I think splitting CAN based GPS for yaw is best considering there are 2 ways to setup in CAN itself. 
One is both modules on same bus and other is having dedicated CAN port for RTCM transfer between the modules. 
The second method requires 2xCAN ports on AP-PERIPH device.

MallikarjunSE, 2022-05-31:
Few pictures I took during tests. If useful.
https://photos.app.goo.gl/zjUpx9qkDYBUtx7y7

closed 2022-11-27 by Hwurzburg