ArduPilot/ardupilot_wiki #4377: Add recommendations for Partner QuadPlane setup

Hwurzburg, 2022-06-03:
@tridge here is a first cut for the wiki....please add/delete/comment

Setup Recommendations for High Reliability SLT/TiltRotor Quadplanes

- Utilize an autopilot with redundant IMUs and sensors
- Utilize an quality airspeed sensor, compass, and GPS. Redudant sensors are suggested, if possible.
- Follow the Wiki's :ref:`plane-configuration-landing-page` and :ref:`quadplane-setup` instructions. Be sure to setup the :ref:`Q_M_BAT_VOLT_MAX<>` param.
- Follow the Wiki's :ref:`quadplane-vtol-tuning-process` instructions before the first hovers. In step 10, using the QuickTune LUA script is preferrable to manual TX tuning or using the AUTOTUNE mode.
- During its first fixed wing flights, autotune using :ref:`automatic-tuning-with-autotune` and :ref:`autocal the airspeed sensor<calibrating-an-airspeed-sensor>`.
- After the first VTOL/FW/VTOL flights, use MAVExplorer's MAGFIT utility to refine the :ref:`compass calibration <using-mavexplorer-s-integrated-magfit-utility>`.

Suggested Parameters (differing from defaults)

- Use :ref:`Q_RTL_MODE<Q_RTL_MODE>` = 3 , to allow all RTL returns to proceed as fixed wing if started at a distance from HOME, then approach and QLAND at HOME
- Use :ref:`Q_OPTIONS<Q_OPTIONS>`, bit 5 = 1 to assure that RC failsafes while in VTOL modes are treated as QRTLs instead of QLANDs
- Set :ref:``Q_ASSIST_SPEED<Q_ASSIST_SPEED>` to be equal to ARSPD_FBW_MIN and that it is 25% above the vehicles stall speed.
- Set the transition failure timer :ref:`Q_TRANS_FAIL<Q_TRANS_FAIL>` to an appropriate time and set :ref:`Q_TRANS_FAIL_ACT<Q_TRANS_FAIL_ACT>` to QRTL
- When setting up the Battery Failsafe, use :ref:`BATT_FS_LOW_ACT<BATT_FS_LOW_ACT>` = 6 for a fixed wing return, loiter down to :ref:`Q_RTL_ALT<Q_RTL_ALT>`, and QLAND
- Although it requires setup per flying location, the use of :ref:`DO_LAND_START<do_land_start>` missions and :ref:`RTL_AUTOLAND<RTL_AUTOLAND>` is advised, to control approach direction.
- Do not set :ref:`RTL_RADIUS<RTL_RADIUS>` too large since the QuadPlane will be in VTOL mode within 2X that radius with most QRTL approaches and 1x that radius on battery failsafes using QLOITER to ALT and QLAND, to minimize battery consumption.

should add something about Weathervaining setup also?

Hwurzburg, 2022-06-24:
@tridge did we want to pursue this?

hendjoshsr71, 2022-08-29:
We should add in an example mission plan for first transition flight test and editing the relevant sitl parameters to test it. 

AKA setup sitl with `--home llon,lat,alt,heading`

Setting up a landing sequence and how much distance to space things based on parameters like `Q_TRANS_DECEL`
Setting up a DO_JUMP for a circuit.

tridge, 2023-01-19:
a few points:
- Q_ASSIST_SPEED should be at least 3m/s below ARSPD_FBW_MIN so we don't kick in during long climbs
- a large RTL_RADIUS is fine, as QRTL will use approach code, which is fixed wing

tridge, 2023-01-19:
planning the DO_LAND_START is an artform, I need to make a video on it
common mistakes:
 - missing a line up WP
 - not using TECS_LAND_ARSDP, it should be set to just above ARSPD_FBW_MIN
 - too short a base leg