ArduPilot/ardupilot_wiki #4392: Search

Pritam2726, 2022-06-20:
Problem:    Pixhawk 2.4.8 Automatically Changing Flight Modes During Flight.

please help me with this problem.

rmackay9, 2022-06-20:
@Pritam2726,

This is probably caused by a failsafe triggering.  Could you post an onboard log in the support forum?

If you're using Copter-4.2.x then this is the best place https://discuss.ardupilot.org/c/arducopter/copter42/161

closed 2022-06-20 by rmackay9

Pritam2726, 2022-06-20:
> @Pritam2726,
> 
> This is probably caused by a failsafe triggering. Could you post an onboard log in the support forum?
> 
> If you're using Copter-4.2.x then this is the best place https://discuss.ardupilot.org/c/arducopter/copter42/161

Thank you for reply
[Hexacopter.zip](https://github.com/ArduPilot/ardupilot_wiki/files/8938289/Hexacopter.zip)

rmackay9, 2022-06-20:
@Pritam2726,

It's very hard to do support on GitHub so could you please open a new topic in the discussion linked above?

Pritam2726, 2022-06-20:
Ok 😊