ArduPilot/ardupilot_wiki #4414: Add DroneID to Peripherals section

rmackay9, 2022-07-01:
On this page we should add a status update on AP support and which hardware is available (or expected to be available and compatible).  We should probably mention which countries the id is valid for.

- https://dronescout.co/dronebeacon-mavlink-remote-id-transponder/  (US?)
- EAMS Robotics has one available for Japan (I will get a link)

BluemarkInnovations, 2022-07-01:
DroneBeacon supports EU, US and also Japan Remote ID standard. (As I understand the Japanese Remote ID standard is equal to the one in USA.)

Also there is a PR for Remote ID support in ArduPilot https://github.com/ArduPilot/ardupilot/pull/21075 (So currently Remote ID transponders are not supported by ArduPilot.)

rmackay9, 2022-07-02:
@BluemarkInnovations,

Great, thanks for that additional info.  I've updated the PR so it lists the regions for the DroneBeacon.  The Japan standard is the same as the US I think although I also wonder if a Japanese "giteki" is required to make it legal in Japan.  Most transmitters require a "giteki".

For now I'm going to assume that PR will get in soon enough that I don't need to add a comment to the wiki..

BluemarkInnovations, 2022-07-04:
Some other things which could be useful for this page:

- free Remote ID Android app: https://play.google.com/store/apps/details?id=org.opendroneid.android_osm
(source code: https://github.com/opendroneid/receiver-android) So users can install this app, to check the information that the Remote ID transponder is broadcasting.
-  I think "giteki" is similar to CE in Europe and FCC in the USA. Although the modules needs to be compliant for each country, ArduPilot also needs to make sure that the module is configured correctly for each region. More info here: https://github.com/opendroneid/opendroneid-core-c#comparison For instance an Operator ID is mandatory in the EU, but optional elsewhere.
- MAVLink capable is similar to saying it has USB. What I don't know is if a MAVLink capable device support the MAVLink messages defined here: https://mavlink.io/en/services/opendroneid.html It could be that devices with GPS (and other sensors) on-board only use proprietary MAVLink messages.

rmackay9, 2022-07-05:
@BluemarkInnovations, thanks for that.  I've added a link to the android app.  I think that AP will provide the mavlink messages on that wiki page although I haven't reviewed it completely.

Hwurzburg, 2022-07-06:
autoclose, reopened for device tracking...

lukasbrchl, 2022-07-07:
Hi, our device Dronetag Mini and Dronetag Beacon supports reading MAVlink messages. We don't use the structures defined in [OpenDroneID](https://mavlink.io/en/services/opendroneid.html). We have described details in our [documentation ](https://help.dronetag.cz/dronetag-mini/mavlink-integration )

friissoren, 2022-07-07:
Just FYI: We are trying to maintain a list of devices capable of transmitting remote ID here: https://github.com/opendroneid/receiver-android/blob/master/transmitter-devices.md

The two lists are probably not completely overlapping, since you will be focusing on add-ons that can feed in MAVLink messages, while the one on the opendroneid page will be more general. I will add a cross link to your list.

rmackay9, 2022-07-07:
@lukasbrchl,

Great, thanks very much for the feedback.  I'll created a [PR to move the Dronetag Mini up](https://github.com/ArduPilot/ardupilot_wiki/pull/4422) into the, "MAVLink enabled devices" section.

rmackay9, 2022-07-07:
@friissoren,

OK thanks!  I've added a link to the OpenDroneId list of devices to the same PR mentioned above.