ArduPilot/ardupilot_wiki #4430: GPS / Non-GPS Transitions

patrickpoirier51, 2022-07-09:
[GPS / Non-GPS Transitions](https://ardupilot.org/plane/docs/common-non-gps-to-gps.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Does this really applies to PLANE ?

Hwurzburg, 2022-07-09:
I dont see why not.....a quadplane approaching and landing in its hanger, for example

patrickpoirier51, 2022-07-09:
OK , your scenario is working.
The problem is users are expecting it to work on plane as of here:
https://discuss.ardupilot.org/t/px4flow-without-lidar-for-non-gps-navigation/87773

Could I suggest You add some additional explanation/restriction as your quadplane example above ?

btw Thanks for your excellent work  Henry @Hwurzburg

Hwurzburg, 2022-07-11:
added note

closed 2022-07-11 by Hwurzburg