ArduPilot/ardupilot_wiki #4446: expo graph update

tridge, 2022-07-18:
The copter and quadplane motor expo section has this graph:
[image]
for large props the graph implies that users should choose an expo of up to 0.9, which is bad advice unless they have good quality thrust stand data to back that up.
We need to modify it so the graph stops at 0.75 - users should not go above 0.75 without good thrust stand data showing the very high expo is needed.

lthall, 2022-07-18:
The line should probably be almost horizontal above 20 inches and drop down to about 0.4 at 4 inches.
For coax aircraft we have very little data but the line could be as much as 0.2 lower for coax.

Hwurzburg, 2022-09-11:
@lthall @tridge  need a new graph for any action on this

closed 2026-09-01 by Hwurzburg