ArduPilot/ardupilot_wiki #4475: docs/raspberry-pi-via-mavlink.rst is outdated

daskalovst, 2022-07-27:
apsync@apsync:~ $ cd /etc/mavlink-router/
-bash: cd: /etc/mavlink-router/: No such file or directory

apsync@apsync:~ $ python3 mavproxy.py --master=/dev/serial0 --baudrate 57600 --aircraft MyCopter
python3: can't open file 'mavproxy.py': [Errno 2] No such file or directory

there are also a lot of miss-aligned data
--master=/dev/ttyAMA0
--master=/dev/serial0

stephendade, 2022-07-28:
Which APSync image are you using? mavlink-router and MAVProxy should already be installed.

>apsync@apsync:~ $ cd /etc/mavlink-router/
-bash: cd: /etc/mavlink-router/: No such file or directory

Instructions for installing mavlink-router are at https://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html#mavlink-router

>apsync@apsync:~ $ python3 mavproxy.py --master=/dev/serial0 --baudrate 57600 --aircraft MyCopter
python3: can't open file 'mavproxy.py': [Errno 2] No such file or directory

Instructions for installing MAVProxy are at https://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html#mavproxy

Hwurzburg, 2022-09-11:
closing, reopen with more information if still an issue

closed 2022-09-11 by Hwurzburg