ArduPilot/ardupilot_wiki #4480: Hazard: Z offset due to gravimetric model EGM96 and solution to this altitude shift

dramida, 2022-08-01:
**A potential hazard in terrain follow and a pain for photogrammetry**

Ublox GNSS used in the majority of the ardupilot builds offer the altitude on WGS84 elipsoid preponderated with the gravimetric model **EGM96** of heights (called geoid) , but SRTM/ Google maps use only **WGS84 elipsoid** heights without gravimetric model overlayed.

There is a significant difference between the two altitudes on the same point of the earth betwen 0 and 100m (because gravity is not uniform, Earth is not omogen inside and the echipotential envelope-mean see level- does not mimic the elipsoid).

This considerable shift is a problemwhen:
- you are geotagging pictures using ardupilot's .bin log and you notice bad Z all over GCP's ;
- you are flying long distances and the gravimetric model changes so the translation of altitude from the initialisation is no longer accurate.
- on further development of the code for next Mars mission :P

Some areas of the planet have more than 20 m gravimetric correction so it exceeds the maximum default Z correction when arming being a potential hazard for terrain follow.

Solution: 
When initializing GPS, set the defaut DATUM/geoid to wgs84, not the mean sea level (EGM96 geoid)

hendjoshsr71, 2022-08-01:
This would probably be better posted on the main repo:

https://github.com/ArduPilot/ardupilot/issues

I was just discussing these models and the default options for GPS's. I think it should be its own parameter instead of hidden in DRV_Options

I doubt we can change the default (at least for old missions or users) without issues if someone were to load an old mission. 

We may need to add a new altitude frame to the Location Class, to allow for missions flown with the desired altitude model/frame.  But any change there would have to have a lot of other necessary cleanups done first.  

Part of OpenDroneID requires that we always have WGS84 altitude.

tridge, 2022-08-01:
the SRTM database we serve on firmware.ardupilot.org is AMSL, not WGS84

Hwurzburg, 2022-08-04:
closing here...open in code issues if further discussion required

closed 2022-08-04 by Hwurzburg