ArduPilot/ardupilot_wiki #4483: RPM Sensor Based Dynamic Notch Setup

adalta87, 2022-08-03:
[RPM Sensor Based Dynamic Notch Setup](https://ardupilot.org/copter/docs/common-rpm-based-notch.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

should there not be a setting to set the INS_HNTCH_MODE to 2 which is rpm sensor. these instructions seem incomplete. After the first bullet should there not be a bullet that states set INS_HNTCH_MODE to 2 which is rpm sensor or possibly 5 which is second rpm sensor?

Hwurzburg, 2022-08-03:
would not hurt but the only way to get  to this page is thru a discussion of that parameter...I'll put it on my list

adalta87, 2022-08-03:
I do not get what you are saying? without setting that parameter you are not correctly setting up rpm based filtering, or am I missing something?

Hwurzburg, 2022-08-03:
the reader would have had to read thru this page in order to access the RPM sensor page....it describes setting the notch mode, then links pages like this for each individual mode type....but it wont hurt to repeat the mode setting on each subpage again

adalta87, 2022-08-03:
well considering I search for rpm based notch filtering on google it took me immediately to that page and not to the first landing page so I do think it needs to be added not going to hurt anything also seems odd the way that page is actually setup why the first landing page then separate for each way. why not just a page each for each filtering option?

Hwurzburg, 2022-08-03:
because duplication of content is a maintenance issue, with 9000 pages of documentation to maintain on a volunteer basis, maintenance is large consideration....there is important common setup information on that landing page that needs to be used in every notch filter...the entire wiki is hierarchical.. any given page is not necessarily complete setup info for a given option...

adalta87, 2022-08-03:
well no matter I think it should properly tell you how to setup RPM filter when you are on the RPM filtering page given that google search do not consider hierarchal structure in their search and I found the RPM filtering page first and not the initial landing page.

Hwurzburg, 2022-08-04:
wiki additions made

closed 2022-08-04 by Hwurzburg