ArduPilot/ardupilot_wiki #4495: Mandatory Hardware Configuration

adalta87, 2022-08-10:
[Mandatory Hardware Configuration](https://ardupilot.org/copter/docs/configuring-hardware.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Where on this page does it include a link to the initial setup feature that is now under mandatory setup, I clicked through the (next) button but never ever came upon a page that shows that feature or explains it? I actually cannot find a page anywhere for this feature and have been seeing it pushed heavily by dev and the likes in the forums. It is also missing from the hierarchal structure that fit with how the wiki is setup. If this is going to be a mandatory setup it should be included in the wiki somewhere.

Hwurzburg, 2022-08-10:
sorry I have no idea of what "initial setup" page you are refering to

Hwurzburg, 2022-09-11:
@adalta87 please provide more information as requested, or I will need to close this issue...thanks

closed 2022-10-09 by adalta87