ArduPilot/ardupilot_wiki #4516: The mode set by Long Failsafe may change when the RC signal is received again.

Rolf-G, 2022-08-18:
I randomly came across that this information about the RC failsafe is unfortunately not correct: 

Wiki: https://ardupilot.org/plane/docs/apms-failsafe-function.html?highlight=failsafe#rc-failsafe-operation

[image]

You can check it very easily on the ground by triggering an RC failsafe by simply turning off the radio transmitter, e.g. in FBWA mode.  If you wait for the long-fale safe and the switch from Circle to RTL and then switch the radio station on again, the mode immediately changes to the mode set on the transmitter when it was switched on. In my example, it immediately switches back to FBWA.

Hwurzburg, 2022-08-18:
This is incorrect as far as I can tell.....both from lots of experience in the air, and just repeated on the bench....same sequence as you state, recovering RC does not change the mode from RTL back to FBWA, only a mode change will move it out RTL....if you change mode on the tx while its off, then turn on, so that the recovered RC mode is different than before, it will change to it...

and it would be a bad thing if it did behave as you state....because intermittant recovery of the RC would switch out of RTL and restart the short fs action, bouncing back and forth until it gets solid RC

Rolf-G, 2022-08-19:
Sorry, you are right. I changed the mode switch while **the transmitter was switched off,** so that the mode was changed when the transmitter was switched on again. However, one should perhaps explicitly warn against carelessly changing the mode on the transmitter when the RC connection is interrupted or if it does happen, that one sets RTL.

closed 2022-08-19 by Rolf-G