ArduPilot/ardupilot_wiki #4537: PRX_ to PRX1_ changes have created warnings in the wiki

rmackay9, 2022-08-25:
[Yesterday's multiple proximity sensor PR](https://github.com/ArduPilot/ardupilot/pull/21521) has broken some links in the wiki
[image]

In general the fix is just to replace PRX_ with PRX1_.  Note that some parameters remain as PRX_ (those that apply to all sensors).

Hwurzburg, 2022-08-25:
a bit more than that, but I will update ASAP

Hwurzburg, 2022-08-25:
shouldnt we add the exclusion and yaw params to the rplidar and terarange tower pages?

Hwurzburg, 2022-08-25:
and something is wrong with your build errors due to this....there are several more pages with errors....that makes me wonder why you do not see them in your test build

Hwurzburg, 2022-08-25:
fixed....let me know about the added params to the other prox sensors pages

closed 2022-08-25 by Hwurzburg

rmackay9, 2022-08-26:
awesome, thanks very much!