ArduPilot/ardupilot_wiki #4553: Tuning Pivot Turns

He202112, 2022-08-30:
[Tuning Pivot Turns](https://ardupilot.org/rover/docs/rover-tuning-pivot-turns.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.
Hi, I have a question to ask I use Rover 4.2.2 and 4.2.3 firmware, set the parameters according to the instructions of the community, and install the motor direction according to the instructions, so that the boat has the function of panning. The following parameters are set: FRAME_TYPE=2,SERVO1—SERVO4_FUNCTION=33,34,35,36. In MANUAL mode, the boat is controlled by the remote control to move forward and backward, turn left and turn right, pan to the left, and pan to the right normally, all of which will use the pivot to run. Plan a straight line, manually adjust the boat to face the first waypoint, switch the Auto mode, the boat can sail automatically, and the heading effect is satisfactory. During the auto sailing process, I found a problem, if the direction of the boat is not facing the next waypoint, switch the Auto mode, the boat will not pivot in place to adjust the direction to go to the first waypoint, it will be faster When turning around with a large radius or leaning to the next waypoint, I set the WP_PIVOT_ANGLE parameter and reduced it from the default 60 degrees to 10 degrees, and the problem has not been solved. After the panning function is turned off, the boat sails normally. How should this problem be solved?
[image]

geofrancis, 2022-08-30:
are you talking about a boat not being able to hold position and pivot turn without moving forwards?

Hwurzburg, 2022-08-30:
@rmackay9 this probably should not be entered here, but hopefully you can help or suggest you can on this support issue

rmackay9, 2022-09-01:
Hi @He202112,

Thanks very much for the report.  OK, so you are right, this appears to be a bug in how we decide whether a vehicle can pivot or not.  I've [added an issue](https://github.com/ArduPilot/ardupilot/issues/21600) and hopefully we can fix this for Rover-4.3.

closed 2022-09-01 by rmackay9