ArduPilot/ardupilot_wiki #4572: AIS (for boats) Value Incorrect

Autonomost, 2022-09-07:
[AIS (for boats)](https://ardupilot.org/rover/docs/common-ais.html) - I tried fixing this and making a PR but looks like this page isn't on master. I just configured a daisy2 yesterday to communicate with ardupilot and this line is incorrect: 

`Type “2” to enable the serial port. “38400 8N1 serial output (NMEA HS w/ adapter)” should be displayed` the value should be 3. 

Also worth noting to make sure users hit `#` and verify their settings are saved when in the serial terminal. Especially after setting the baud rate.

If someone points me to the branch I'll make all these changes and push em back out. Thanks!

Hwurzburg, 2022-09-07:
thanks...the doc is in master, just some pages dont want to use the web editor....you can fork and clone the repo and make changes, push back to your git repo  and create a PR...or I can do it , if you arent familiar with local repos and creating PRs...let me know

Autonomost, 2022-09-07:
sounds good. PR inbound shortly. Thanks!

closed 2022-09-11 by Hwurzburg