ArduPilot/ardupilot_wiki #4672: wiki needs a dedicated page for each and EVERY board that ardupilot runs on -  autogenerated from hwdef plus some data source we don't have yet..?

davidbuzz, 2022-10-18:
Should include at a minimum?:
  - one picture of the flight-controller-pcb as-shipped, with user-labels on PCB/ silkscreen readable.  
  - If there are label/s for pins/connectors on both sides of the pcb, it should include a picture of each side of the pcb.
 - for flight controller/s shipped in a case, it should include enough pictures of the case that all external label/s can be read/identified.
- a table/list of all readable "labels" a user would typically encounter when using the board, eg "UART 1", "TELEM2", "D2", "T5", "R6", as seen in the above picture/s, and what ardupilot's internal software name for this connector is. eg "UART1 -> Serial 4", or "D4 -> GPIO53"  
- as many 'specs' as can automatically be derived from the hwdef.dat, such as processor type/speed, onboard sensors list, number of GPIO ports, number of CAN ports, number of analog-inputs..? 

Optionally might want ?:
- a pinout list for each "port"
- a "freetext" manufacturer-specs/system-features block
- external manufacturer url/s for , eg, their website, store, resellers, forums, etc.
- link to schematic/s, if available.
These page/s should optimally, be automatically generated from existing data source/s. I know we don't have all these data sources available right now, in particular the picture/s and user-labels aren't in hwdef.dat right now, so I suspect we'll need to  agree that this is a good idea, and add some more support to the hwdef.data parser to allow this extra metadata to be stored either in the hwdef.dat, or in another repo that parses hwdef.dat files and adds images and labels.

I'm generally thinking that this auto-generated output should [in the future] be able to replace , for example these sorts of [manually curated right now] page/s:
https://ardupilot.org/copter/docs/common-durandal-overview.html
https://ardupilot.org/copter/docs/common-cuav-v5plus-overview.html

it would be nice if we could also make the source data parsable enough to improve our parameter xml, not sure exactly whats needed there, but we can start collecting pics/labels/data toward this goal without knowing how the xml will be handled, so long as the wiki-page we auto-generate is able to show pics/labels/relationships to the user, so the user doesn't need to read the hwdef.data to assemble and setup a vehicle.

davidbuzz, 2022-10-18:
@peterbarker  @tridge  @Hwurzburg

davidbuzz, 2022-10-18:
here's a bit of kinda throw-away python code that  parses all the hwdef.dat files, and pulls out a few things that might be relevant to building a wiki page with it.... for now it just emits the result as json... but its the sort of skeleton i was thinking of..
https://gist.github.com/davidbuzz/338f52ee9513f2297d08afe4b7bc4d98
```
...some output...

 "BeastH7": {
        "images": {
            "beast_h7_pinout.png": "ardupilot/libraries/AP_HAL_ChibiOS/hwdef/BeastH7/beast_h7_pinout.png"
        },
        "mcu": "STM32H7xx STM32H743xx",
        "FLASH_SIZE_KB": "2048",
        "SERIAL_ORDER": [
            "OTG1",
            "USART1",
            "USART2",
            "USART3",
            "UART4",
            "EMPTY",
            "EMPTY",
            "UART7"
        ],
        "GPIO-PD15": "80",
...  ```

Hwurzburg, 2022-10-18:
I am 100% in favor of anything that can be autogenerated in the wiki.
I spend 3-4 hours creating every new board page.
Although we now usually require a README file for every new board definition file, it is not thoroughly reviewed for content or accuracy before merging. Most are inaccurate and incomplete. I spend a lot of time checking the actual hwdef file. In addition, I usually have to graphically edit the board photos, if provided, as well as supplement them. I also usually have to create pin map tables to match the board pinouts and add missing information. I also usually have to verify if default.parm files correctly setup the boards in those baords that are usually migrated from Betaflight or INAV.
Any autogenerated wiki page MUST be whole and complete in and of itself...there is no mechanism that would allow individual modifications. All existing autodocs are very focussed containing single source info: LOG messages, param lists, build file build options
Any board page must have the following sections:
1. Board Title: this often does not correspond to the hwdef file name. In addition, many boards with differing chararcteristics can use the same hwdef (different carrier board implementations like the Pixhawk6X). Some hwdefs do not correspond to any board but are base includes or are private creations or just currently wrong like the SkyStars board def(these could be handled with whitelists).
2. Board Photo: these usually are edited by me or are missing from the README, or not a good photo
3. Description of the board's major features including its unique features
4. Where to buy: a link the the OEMs store page for the board
5. Specification list: lists cpu, sensors,interfaces,bd dimensions, power, etc.
6. Pinout or Wiring Diagram: either diagrams showing peripheral connections to the board or a list of board pins/connectors with pin descriptions
7. UART table: a map of SERIAL->UART->default protocol + DMA capability, plus info on missing pinouts and alternate board configuration of UARTS
8. RC Input Section: Description of how RX is attached. RCIN interrupt vs UART and alternate board configurations, if any
9. PWM Outputs: Which outputs belong to which timer groups, which outputs are DShot or BDShot capable Plus. Also if an output is setup for LED use by defautl
10. Battery Monitoring: What type of battery monitor is setup. Some have internal, some external, some CAN, some analog. Some need special caps to prevent noise injection into the current sense
11. GPIOS: While a list would be nice, many dont include them, but if they have special functions like VTX power or Camera control, they are listed...you have to decode the hwdef comment fields sometimes.
12. RSSI: if an analog pin is designated for rssi input or airspeed input
13. Firmware: Which build target corresponds to this board...again, some boards have the same target with different board names...some boards do not have the same name as the target

In addition, sometimes the boards have integrated peripherals that would not be part of the hwdef, so they have to be described and included.

All of these would need to be autogenerated in the proper display format and completely since there is no way to make individual tweaks to a given page.

I do not believe this is realizable. However, improving the documentation in the hwdef folder would be a big help and does not need to be either complete or a wiki page format. This would make the task for coders for generating a README easier and allow me not to have to  recheck everything in the README against the hwdef and defaults.parm.

So an additional file autogenerated that extracted the following from the HWDEF and defaults.parm would be helpful and avoid me constatntly flipping back to the HWDEF to verify:
1. SERIAL->UART with DMA  and if any pins are not defined, any changes from default setup by the defaults.parm file
2. A list of any line with a GPIO label and its preceeding comment field, if any
3. Battery Monitor setups if present
4. is an RCIN interrupt pin defined or not
5. Any alternate board configs
6. PWM timer groups

davidbuzz, 2022-10-20:
It sounds to me like everyone is in-favour of improving the meta-data in the hwdef area [and related data such as board pics] in the same folder/s .. if we can get it good-enough to auto-generate wiki pages from in the future, then that's great, but it doesn't need to be the only goal.... just having more complete meta-data for us to use as reference material for hand-curated wiki pages is certainly still a step in the right diurection.,, but we *might* be able to automate more of it in the future.

Goal 1: define a good set of meta-data that we can start collecting toward this goal, and as an interim, wiki-doc people can optionally/choose to use this as reference material when building a board-documentation page...  eg board pics, wiring diagrams, manufacturer urls, etc .   
Goal 2:  Decide what method works best/easiest for storing/collecting this data.. eg  on-disk in an existing git repo[ardupilot or wiki or new], or in a sqlite database?,  a web folder with a bunch of JSON in it? a S3 bucket? a simple web tool to make this easy to edit?,  and implement at least a minimum-viable process for making editing of it trivially easy for wiki-maintainers and board-maintainers.
Goal3: do a once-off populate of as much data as we already have - getting the admittedly-incomplete data we already have as a starting point
 - For the board/s where we already have great static documentation[ie data from wiki repo], we could copy-paste a lot of the content from these into the Tool made in Goal 2... or better yet, write some throw-away python to do it.
 -  For the data coming from hwdef file/s [data from ardupilot repo], and the related meta/data do something similar.... this script should probably be non-throw-away, as we'll want to be able to re-run this whenever a new board comes into existance in the ardupilot repo
Goal 4: write a python script that assembles the collected data into a set of generated wiki pages, one for each board, so we can review and see shortcomings etc.
Goal5: if/when it good-enough to replace the static content that we have there now [ie when wiki-master is happy] , integrate it with the wiki-publish process.
thoughts?

Hwurzburg, 2026-09-13:
@davidbuzz I think we can close this now....long ago we decided that we would improve the quality of the hwdef before merge and simply point to that from the wiki, which we have been doing since 4.6....I have a review bot that greatly reduces the review/re-review time for the hwdef from a applicatons/doc consistency point of view and Andy checks it for ChibiOS correctiness.....the bd review backlog has effectively been eliminated at this point...all pending boards are awaiting changes be made from reviews or waiting on CI  fixes or Devcall consideration....so I will close this

closed 2026-09-13 by Hwurzburg