ArduPilot/ardupilot_wiki #4684: Here  RTK GPS

dmitrievivan, 2022-10-26:
Tell me where to get the source code of programs for HERO V2 or at least exchange protocols

rmackay9, 2022-10-27:
Hi @dmitrievivan,

So I guess you're asking about the interface to provide RTCM data to ArduPilot?  I'm not personally very familiar with how this works but in general, the data arrives via mavlink in either in the [GPS_RTCM_DATA](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L5289) or [GPS_INJECT_DATA](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L4885) messages and this is then passed down into the various GPSs that can support RTK (UBlox, NMEA, UAVCAN).

I did a search for RTCM in our code and various things turned up.  For example [AP's UBlox driver is here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_UBLOX.h) (which is what the Here GPSs (and many others) are based on.

Hwurzburg, 2023-11-30:
answered and stale...closing

closed 2023-11-30 by Hwurzburg