ArduPilot/ardupilot_wiki #4687: Document how companies can download logs from autopilot to a flight controller

rmackay9, 2022-10-27:
We should document (perhaps on the developer wiki) how companion computers can download onboard logs from the autopilot.  The two methods I know of are:

- download the logs via MAVLink just like Mission Planner and MAVProxy do
- install dflogger on the companion computer and configure AP to stream the onboard logs to the companion computer
- MAVFTP the log file off the autopilot (not recommended)

Williangalvani, 2022-11-16:
> install dflogger on the companion computer and configure AP to stream the onboard logs to the companion computer

Is [this](https://github.com/peterbarker/dronekit-la/blob/master/dataflash_logger.cpp) dflogger?

I 'm wondering if this would be a good solution for https://github.com/bluerobotics/BlueOS-docker/issues/457

khancyr, 2022-11-16:
There is also mavlink-router to get the logger streamed.

Or a simple pymavlink script to retrieve the log

closed 2026-08-25 by Hwurzburg