ArduPilot/ardupilot_wiki #4702: Setting Motor Range

PAPYBULL, 2022-11-05:
[Setting Motor Range](https://ardupilot.org/copter/docs/set-motor-range.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.
Measuring the deadzone
After carrying out the ESC calibration according to “All-at-once” method, I proceeded to the testing of my quadrotor vehicle. That trial revealed that one of the four motors doesn’t start at the same time than the others. Therefore, it seemed obvious that the corresponding ESC was subject to a dead zone which should be measured and suppressed as recommended in Arducopter’s documentation.
According to those instructions i.e., Arducopter >First Time Setup>Mandatory Hardware Configuration>Setting Motor Range, one should refer to Mission Planner's Initial Setup (which goes beyond the scope of Arducopter).
After opening the page and selecting the Optional Hardware menu, a Motor Test submenu was supposed to appear, at the bottom of the list, to check the characteristics of the 4 motors A, B, C, D.
Unfortunately, this Motor Test submenu did not appear, in my case! Why?
Consequently, the dead zone problem cannot be solved by that way in my case! What do you suggest?

Hwurzburg, 2022-11-05:
this should be moved to a support page in discuss.ardupilot.org, probably under the Copter section....this is for documentation issues and this is a setup or Mission Planner issue...thanks

closed 2022-11-05 by Hwurzburg

PAPYBULL, 2022-11-05:
Thanks for your reply. Sorry as I just subscribed for being a member since
yesterday, I am not familiar with the procedure. So I failed to move the
text as you suggested

Hwurzburg, 2022-11-05:
Welcome! No problem