ArduPilot/ardupilot_wiki #4731: Setting up the Build Environment on Windows 10/11 using WSL1 or WSL2

adalta87, 2022-11-20:
[Setting up the Build Environment on Windows 10/11 using WSL1 or WSL2](https://ardupilot.org/dev/docs/building-setup-windows10.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

I am working on building this on a new laptop and all things go well but step 4 seems kind of out of place and lacking much info. where does this fit into building the code? Is this really a step needed to compile the code? can someone please explain. @rmackay9 I had read you were working on WSL for Win11 or remember reading somewhere that someone was updating the docs on that.

Hwurzburg, 2022-11-20:
on Win 10 VxSvr or other is needed for any app needing GUI (MAVProxy,MAVExplorer)....Win11 integrates this and its not needed....I think maybe just a note anytime XWindows is mentioned that its not needed for Win11 might be all thats required...
@rmackay9 was working on getting instructions on how to setup VSCode environment, but has not completed those yet to my understanding....what do you think about just adding notes to that page about Win11 not needing that step?

adalta87, 2022-11-20:
I am very new to a lot of this so I am just trying to warp my head around where this fits into me being able to do sim using a win11 PC using WSL that I have downloaded ubuntu 20.04.5 LTS from the microsoft store. I am able to compile the code just fine but I get confused about step 4 and where I need to go next for sim using the WSL. Sorry if this doesn't make sense.

Hwurzburg, 2022-11-20:
@TunaLobster what do you think about adding the note I suggest....as a stopgap for now until Randy finishes?

TunaLobster, 2022-11-20:
Seems like a good idea. Just a "Windows 10 only" at the start of step 4?

rmackay9, 2022-11-20:
Adding that line to step4 sounds good to me.  I hope to eventually get back to improving that page but I can't say when sadly.

Hwurzburg, 2022-11-27:
note added...closing this

closed 2022-11-27 by Hwurzburg