ArduPilot/ardupilot_wiki #4769: Cartographer SLAM for Non-GPS Navigation

hbhskf, 2022-12-02:
[Cartographer SLAM for Non-GPS Navigation](https://ardupilot.org/dev/docs/ros-cartographer-slam.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Now,I have completed all the operations according to the content of the URL, and most of the information is correct, but in mavlink_inspector, I cannot see the information of OBSTACLE_DISTANCE, and there is no distance information of obstacles in Proximity.

Device: rplidar S2,pixhawk 2.4.8,Jetson Nano
firmware:FMU V3,  Rover 4.1

Hwurzburg, 2022-12-02:
@snktshrma can you please respond to this issue?

snktshrma, 2022-12-02:
Hi @hbhskf! So OBSTACLE_DISTANCE is removed in [this commit](https://github.com/ArduPilot/ardupilot/commit/a56ab9136ee7c7a2211dc793510c1019a8fa6722)
Still these messages are being accepted by the FCU. So it's ok to go on.

hbhskf, 2022-12-03:
> 

OK thank you

closed 2026-08-28 by Hwurzburg